nimblemachines/muforth #14: chipdef files state that it's contents are extracted from the datasheets, but the scripts used to do so seem to be missing

ForceFaction, 2020-08-28:
Hi, 
the chidefs.lua files all state, that the data got extracted with lua, but the actual scripts aren't there and there is no comment on where to find them. 

Including at least a reference would make it way easier to create chipdefs, for new controllers.

nimblemachines, 2020-09-25:
The `chipdefs.lua` files are a combination of hand-scraped strings (ie, copied and pasted, using a mouse, from the PDF datasheets), together with some small functions that parse those strings and spew out Forth code.

In every case a chipdefs file starts with named strings (using Lua's [[ ]] quotation) and ends with the parsing and printing functions.

Adding a new chip to a chipdefs file and generating muforth code from the chipdefs file both require manual intervention and occasional tweaking.

In order to add a new chip, I usually add a new named string containing the text from the datasheet that describes all the registers for that chip. Then I add or modify one of the parsing and printing routines, and then iteratively execute the chipdefs file, using a command line something like this:

```
cd muforth/mu
lua target/MSP430/device/chipdefs.lua
```

Usually I'll pipe the result to `less` until I know it's working, then I'll redirect to a file. The "problem" is that the existing chipdefs scripts generate the contents of multiple files and simply concatenate them. I have to then un-concatenate them in a text editor and paste the results where they are needed.

Sometimes I'll make a change to how the (generated) muforth code is formatted, and in that case I run the Lua file (as above) and copy/paste the generated contents into all the requisite files, and then commit what has changed.

Since I rarely make changes to any of the chip equates files, this manual process hasn't been a problem. But if there were many changes being made or many files being generated, finding a way to make the process more automatic would make sense. The existing setup is also harder for others to contribute to.

Out of curiosity, which target or targets are you interested in adding new chipdefs to?

Also, if you are interested in ARM targets, you should take a look at two other projects of mine: [kinetis-chip-equates](https://github.com/nimblemachines/kinetis-chip-equates) for Kinetis devices, and [stm32-chip-equates](https://github.com/nimblemachines/stm32-chip-equates) for STM32 devices. Those are works-in-progress too, and none of the generated equates files have yet been incorporated into the muforth source tree.

ForceFaction, 2020-09-28:
Thanks for the explanation :)

i was trying to get muforth on an atmega8-16pu. I'm relatively new to forth in general so i also am rather slow in understanding how i should use muforth :D The only thing i found (which seems to use a controller compatible with the one i want to use is https://yellowrabbit.gitlab.io/blog/2019/diy-ps_2-adapter/)

Some more documentation or some small example projects would be awesome :)