nimblemachines/muforth #42: Longan Nano and DFU Flashing

anarchitech, 2021-12-18:
Per commit a0750943f78cb8f9f55d687086200220a6fa1efc we appear to have joy with DFU flashing of the Longan Nano using a USB-C cable (as opposed to the serial pins):

```
sid@vicious:~/muap/mu$ ./muforth -d bootloader -f target/RISC-V/board/longan-nano.mu4 -f target/RISC-V/load-chat.mu4 
muforth/64 (48ef1d57) 2021-dec-15 07:33 (https://muforth.nimblemachines.com/)
Copyright (c) 2002-2021 David Frech (read the LICENSE for details)

Type 'settings' to see a few of muforth's tweakable behaviours.

(( Sipeed Longan Nano board 
(( GigaDevice GD32VF103CBT6 chip 
(( GigaDevice GD32VF103xB equates ))
(( Target endianness ))
(( Meta-compiler (chains and token consumers) ))
(( RISC-V memory image ))
(( RISC-V RV32 assembler (I subset) ))
(( RISC-V RV32 assembler (C subset) ))
(( RISC-V RV32 disassembler (I subset) ))
(( RISC-V RV32 disassembler (C subset) ))
(( RISC-V meta-compiler (main) ; again.  ))
(( RISC-V interaction dumps again.  ))
(( Flash programming ))))
(( Serial expect/send ))
(( STM32 serial bootloader support ))
(( STM32 USB DFU support ms again.  >cmd again.  erase-page again.  setup-chunk again.  ))
(( RISC-V serial chat protocol (host) ?spkt again.  >b again.  b> again.  ))))
(( RISC-V serial chat protocol (core) ))
dfu 
WARNING: Target's flash image differs from host's. Run flash-image? Ok (chatting) (hex)
flash-image 
erase   800_0000 
program 800_0000 36c  Ok (chatting) (hex)
verify  Ok (chatting) (hex)
```

However there appears to be an **issue reconnecting with the target over USB**.  The procedure given in commit message a76d40267ff804fe07d54073a646256146c3c54c 

+ Omits mentioning that dfu should be substituted for boot when flashing via that method
```
     ./muforth -d bootloader -f target/RISC-V/board/longan-nano.mu4 \
            -f target/RISC-V/load-chat.mu4
       boot        
       flash-image
       verify
       ^C
```
+ and further gives a procedure for reconnecting to the target and verifying the flash procedure ...
```
    You should now reset the board (by pressing the reset button), re-connect,
    and verify that everything worked:
    
      ./muforth -f target/RISC-V/board/longan-nano.mu4 \
            -f target/RISC-V/load-chat.mu4
      chat
      verify
```

... but this  reconnect procedure fails and the observed behavior on ChromeOS, Archlinux and OpenBSD is that while attempting to chat with the target via USB-C cable:

+ The target **fails to enumerate** on the host without first holding down reset, pressing boot, releasing reset and releasing boot

+ **Fails to chat** with the target and verify ...

```
sid@vicious:~/muap/mu$ ./muforth -f target/RISC-V/board/longan-nano.mu4 -f target/RISC-V/load-chat.mu4 
muforth/64 (48ef1d57) 2021-dec-15 07:33 (https://muforth.nimblemachines.com/)
Copyright (c) 2002-2021 David Frech (read the LICENSE for details)

Type 'settings' to see a few of muforth's tweakable behaviours.

(( Sipeed Longan Nano board 
(( GigaDevice GD32VF103CBT6 chip 
(( GigaDevice GD32VF103xB equates ))
(( Target endianness ))
(( Meta-compiler (chains and token consumers) ))
(( RISC-V memory image ))
(( RISC-V RV32 assembler (I subset) ))
(( RISC-V RV32 assembler (C subset) ))
(( RISC-V RV32 disassembler (I subset) ))
(( RISC-V RV32 disassembler (C subset) ))
(( RISC-V meta-compiler (main) ; again.  ))
(( RISC-V interaction dumps again.  ))
(( Flash programming ))))
(( Serial expect/send ))
(( RISC-V serial chat protocol (host) ))))
(( RISC-V serial chat protocol (core) ))
chat 
chat No such file or directory
```
+ ... **unless the -d bootloader switch is included and dfu is executed**:
```
sid@vicious:~/muap/mu$ ./muforth -d bootloader -f target/RISC-V/board/longan-nano.mu4 -f target/RISC-V/load-chat.mu4 
muforth/64 (48ef1d57) 2021-dec-15 07:33 (https://muforth.nimblemachines.com/)
Copyright (c) 2002-2021 David Frech (read the LICENSE for details)

Type 'settings' to see a few of muforth's tweakable behaviours.

(( Sipeed Longan Nano board 
(( GigaDevice GD32VF103CBT6 chip 
(( GigaDevice GD32VF103xB equates ))
(( Target endianness ))
(( Meta-compiler (chains and token consumers) ))
(( RISC-V memory image ))
(( RISC-V RV32 assembler (I subset) ))
(( RISC-V RV32 assembler (C subset) ))
(( RISC-V RV32 disassembler (I subset) ))
(( RISC-V RV32 disassembler (C subset) ))
(( RISC-V meta-compiler (main) ; again.  ))
(( RISC-V interaction dumps again.  ))
(( Flash programming ))))
(( Serial expect/send ))
(( STM32 serial bootloader support ))
(( STM32 USB DFU support ms again.  >cmd again.  erase-page again.  setup-chunk again.  ))
(( RISC-V serial chat protocol (host) ?spkt again.  >b again.  b> again.  ))))
(( RISC-V serial chat protocol (core) ))
dfu  Ok (chatting) (hex)
verify  Ok (chatting) (hex)
```
### But ... Serial Joy (PL2303)

However, **if we connect to the target via the serial pins** (in this case with a PL2303) we note that everything Just Works post DFU flash:

```
sid@vicious:~/muap/mu$ ./muforth -f target/RISC-V/board/longan-nano.mu4 -f target/RISC-V/load-chat.mu4 
muforth/64 (48ef1d57) 2021-dec-15 07:33 (https://muforth.nimblemachines.com/)
Copyright (c) 2002-2021 David Frech (read the LICENSE for details)

Type 'settings' to see a few of muforth's tweakable behaviours.

(( Sipeed Longan Nano board 
(( GigaDevice GD32VF103CBT6 chip 
(( GigaDevice GD32VF103xB equates ))
(( Target endianness ))
(( Meta-compiler (chains and token consumers) ))
(( RISC-V memory image ))
(( RISC-V RV32 assembler (I subset) ))
(( RISC-V RV32 assembler (C subset) ))
(( RISC-V RV32 disassembler (I subset) ))
(( RISC-V RV32 disassembler (C subset) ))
(( RISC-V meta-compiler (main) ; again.  ))
(( RISC-V interaction dumps again.  ))
(( Flash programming ))))
(( Serial expect/send ))
(( RISC-V serial chat protocol (host) ))))
(( RISC-V serial chat protocol (core) ))
chat 
Chat firmware version 48ef1d57 Ok (chatting) (hex)
verify  Ok (chatting) (hex)
```

nimblemachines, 2021-12-18:
Other than the misleading commit message in a76d40267ff804fe07d54073a646256146c3c54c, the behavior you are seeing is as expected - and intended.

In order to be able to "chat" with the target over USB, I have to write USB firmware for the device.

The problem: 

GigaDevice decided to embed the very complicated Host/OTG/Device USB peripheral from the STM32F105/107 rather than the much simpler USB fullspeed peripheral in the F07x, F103, F303, etc.

It's on my list to write USB firmware for all these devices, but I'm going to start with the easier ones.

Until then, while it's possible to **flash** the Nano (and other GD32VF103 based boards) with the DFU over USB, you have to switch to the serial connection to **chat** with the device.

Here is an edited, and hopefully less misleading, set of instructions for installing or updating the **serial** chat code on the Longan Nano:

### Install or update via serial bootloader

Connect a [USB-serial adapter](https://muforth.nimblemachines.com/using-usb-serial-adapters/) to the four serial port pins on the end of your Longan Nano. Make sure `serial-target` points to your USB serial adapter.

Reboot your board into the serial bootloader: Hold down reset, press boot, release reset, release boot. Then do this:

```
./muforth -d bootloader -f target/RISC-V/board/longan-nano.mu4 -f target/RISC-V/load-chat.mu4
boot
flash-image
verify
^C
```

### Install or update via DFU over USB

Connect the Nano to your computer using a type-C USB cable.

Reboot your board into the DFU bootloader: Hold down reset, press boot, release reset, release boot. Then do this:

```
./muforth -d bootloader -f target/RISC-V/board/longan-nano.mu4 -f target/RISC-V/load-chat.mu4
dfu
flash-image
verify
^C
```

Follow the instructions below to verify, via serial chat, that it worked.

### Test the installed serial chat code
 
Whether you installed via DFU or the serial bootloader, you have to connect via serial to chat with the board. Again, make sure that `serial-target` points to your USB-serial adapter.

Reset the board (by pressing the reset button), re-connect, and verify that everything worked:

```
./muforth -f target/RISC-V/board/longan-nano.mu4 -f target/RISC-V/load-chat.mu4
chat
verify
````

anarchitech, 2021-12-18:
Great, this is what's required.  My intention is to roll this into a dedicated riscv-target
README on my fork, what do you think about doing that on master? The sooner we can start extracting useful info like this out of github and issue messages and into some dedicated documentation, the better.  It doesn't have to be perfect, it just has to start and then get beaten into shape along the way.

anarchitech, 2022-10-23:
Closing this as I think we've covered it.

closed 2022-10-23 by anarchitech