nimblemachines/muforth #47: Mecrisp Ice / J1 FPGA Port

PythonLinks, 2025-02-04:
This repository was recommended on the Forth Discord Server. 

I am working with the J1/Mecrisp Ice Forth Family on FPGAs.  These are 16 bit processors, with an address space of 8K words, of which usually about 5 K words are for the Forth libraries. 

The competitive advantage of Forth over RISC-V soft cores on small devices is size, speed and memory utilization.  A real market nice for Forth.  To be able to further shrink the memory required would be awesom. 

> This makes it possible to target very small devices that only have a few kilobytes of flash and a few hundred bytes of RAM. In contrast, a self-hosted Forth often needs 16 KiB (or more) of flash to be useful, and consumes RAM for the text interpreter and compiler.

So I am new to this, and a bit lost.  You asked for documentation feedback.  I am lost at the high level concepts.  Even before I get to startup.mu4.  Will I be running MuForth on the J1?  Will I be porting Mecrisp-Ice to MuForth? 

If I am running MuForth on the J1, what words do I have to implement on the J1? 

If I have to port Mecrisp-Ice Forth to the MuForth platform, oh boy does that sound scary.  Here is the Mecrisp Ice cross compiler documentation which I wrote. 

https://mecrisp-ice.readthedocs.io/en/latest/cross-compiler.html

In due course, I will figure all of this out.  This is just my feedback to you after looking at MuForth for 5 minutes.  So that you have a better idea of the questions which newbies will ask. 

I really like your concept of conviviality.

PythonLinks, 2025-02-04:
The original README, 
https://muforth.dev/readme/
has a great description of the cross compiler.  
that should be front and center in the current readme.  It gives the high level view of what is happening. 

Mecrisp Ice does two interesting things.  One, there is a bit in the ALU instructions for return.  No need for an extra instruction. 

And the other is that the dictionary is more complex.  
This is a 16 bit soft core.  Three bits define the instruction type, leaving 
13 bits for a call or jump address.  The usable address space is 13 bits. 
So 13 bits in the dictionary point to the previous entry in the dictionary. 
The other 3 bits are flags for foldability.  

Then there is another byte.  The first bit is if it is an immediate word.  The next 7 bits are for the length of the word.  Then the ascii name of the word. 
Then the executable code.  The last executable word in an ALU word with the return bit set. 

The code implements tail recursion. 

There is also a difference between how FPGAs and micro controllers work.  You mentioned twig and leaf code.  On a microcontroller the leaf code implements various forth words in assembler.  On an FPGA, there is no leaf code.  Low level Forth words are directly executed by the hardware.   So the compiler checks the dictionary.  If the implementation of a called Forth word is a single machine instruction, then that instruction gets inlined into the calling word's executable code. 

Maybe the mecrisp-ice documentation should have a glossary of words implemented in hardware.

PythonLinks, 2025-02-04:
Discussion of MuForth on the Discord Forth Hardware Channel. 

https://discord.com/channels/780612541013033030/1304408536687841353/1336053864754970695

Here is a 7 day  invite to that server. 
https://discord.gg/mjz4tkEc

nimblemachines, 2025-02-05:
The J1 is a neat design. I was at the Forth meetup many years ago when James Bowman introduced it!

While I am a huge fan of using FPGAs to play around with CPU designs (especially for Forth and RISC-V processors), with muforth I have been focusing on (a handful of) mainstream microcontrollers, for two main reasons:

* they have lots of useful peripherals
* they are inexpensive and easily available

Even if there is extra space available in your FPGA after synthesizing your CPU, are you going to have the time and skill to create timers, UARTs, SPI interfaces, etc? And even if it's feasible to do *that*, it's impossible to put analog i/o on your chip (but maybe that doesn't matter).

Whereas even a $2 PIC18 will have a lot of useful analog and digital i/o on it, and, with a bit of coaxing, it will run Forth (or something that looks a lot like it).

Unfortunately, getting muforth to work with a new target - writing the assembler and compiler to generate the target code, and developing a way to chat with the chip - is non-trivial. In the case of the J1, you would need to adapt either Bowman's swapforth or mecrisp ice - at least the pieces that generate the code - and then write your own debug stub ("chat" program) to talk to the host to make the whole experience interactive (the chat program is what makes the "tethered" aspect of muforth work).

The existing targets are good sources of examples of how to do this, and the code therein is generally well-documented, but there isn't a good high-level overview of what is involved and how the pieces fit together. Yes, I really need to write this!